Dlloydev/sTune #1: How to use sTune. (Doc issue? PBCAK?)

robcazzaro, 2022-01-04:
I just ported a project from the old PID_v1 library to QuickPID, and everything works well (thanks for all your work!) But my system in unstable (was unstable even with PID_v1), so I want to use sTune.

I looked at the examples for sTune, in particular the QuickPID one, and I'm struggling to understand when the code exits, so that I can retrieve the Kp, Ki and Kd parameters for future use. I looked at the Run method, and I see a similar state machine with some of the same values used in the switch statement in the example (inOut, tunings and runPid). It looks as if the loop() will keep running forever

I'm pretty sure I'm missing something (https://acronyms.thefreedictionary.com/PBCAK), but I think that a couple of extra explanation lines in the readme would help other people as well

Dlloydev, 2022-01-04:
Thanks for trying sTune! 

Yes, I agree the documentation is quite light at this point ... more will be added soon.

> I'm struggling to understand when the code exits, so that I can retrieve the Kp, Ki and Kd parameters for future use.

I've added some comments to the loop code (shown below). Note that the whole process is non-blocking and the Kp, Ki and Kd variables in the sketch get automatically updated by sTune (case tunings). You can also use the GetKp(), GetKi() and GetKd() functions in your code.

```c++
void loop() {
  switch (tuner.Run()) {  // active while sTune is testing (non-blocking)
    case tuner.inOut:
      Input = (analogRead(inputPin) / mvResolution) - bias;
      analogWrite(outputPin, Output);
      break;

    case tuner.tunings:                        // active just once when sTune is done
      tuner.SetAutoTunings(&Kp, &Ki, &Kd);     // sketch tuning variables are updated by sTune
      myPID.SetMode(myPID.Control::automatic); // the PID is turned on (automatic)
      myPID.SetSampleTimeUs((testTimeSec * 1000000) / samples); // PID sample rate
      myPID.SetTunings(Kp, Ki, Kd);            // update PID with the new tunings
      break;

    case tuner.runPid:  // this case runs indefinitely after case "tunings" (non-blocking)
      Input = (analogRead(inputPin) / mvResolution) - bias;
      myPID.Compute();
      analogWrite(outputPin, Output);
      break;
  }
  // put your main code here, to run repeatedly:
}
```
Note that the PID sample rate should be whatever you use (and is ideal) for your system.

```c++
myPID.SetSampleTimeUs(100000);  // 100ms, 0.1sec
```

robcazzaro, 2022-01-04:
This is perfect, thanks!

The added comments are super-helpful in clarifying the logic and how sTune could be used even on the final sketch. For my use, though, I would prefer to run sTune initially to figure out the best parameters, then create a final sketch with only the final parameters and QuickPID

But I can see a project where the physical conditions change frequently enough where having a menu-driver autotune makes sense (for example a sous vide with different size containers). In that case, your example program would become the starting point for the final sketch.

Thanks again for writing/sharing your code and the prompt reply! I'm waiting for the final component to arrive, then I will be able to run sTune and report any interesting finding. I'm leaving this issue open for now (but feel free to close if you prefer)

Dlloydev, 2022-01-04:
> For my use, though, I would prefer to run sTune initially to figure out the best parameters, then create a final sketch with only the final parameters and QuickPID

- Next version I'll add an example that can report tuning parameters based on all 10 tuning rules after running just 1 test.
- Also a brief tutorial or example on using sTune print data c/w tuning parameters in [pidtuner.com](https://pidtuner.com/#/). This on-line tuner can identify the PID model, allow fine-tuning the parameters then instantly seeing the results on plots.

No problem leaving this open (It'll be my reminder for the next revision). 

Looking forward to your findings!

robcazzaro, 2022-01-07:
I got sTune to work for my setup. It works well (apart from an issue I just logged), and the current documentation is really good.

The only additional suggestion I would have is to explicitly explain that the algorithm starts from an output value that should result in a steady state system, and creates a step response by increasing the output to a different value.

In the case of a heating system, it's difficult to find an output value that creates a steady state. For example, in my case, with an output 0f 10% PWM (for an SSR controlling a heating element), I get to a quasi steady state of ~90F, but it keeps slowly drifting upwards even after 5 minutes of settling

If I were to use, say, 40% PWM, the heater will become too hot for its specs and never reach a steady state before getting damaged

I can still autotune my system, simply have to use lower starting points. I think that making it more clear in the documentation would help other people dealing with slowly heating systems where the heating element is much more powerful than needed for small temperature changes

Dlloydev, 2022-01-07:
Thanks for the feedback and suggestions ... glad you've had some success!

The examples I have so far were only to check the operating range of sTune. (there's many more left to do). The closest example to what you're describing (difficulty in getting to steady state input) is [Step-Test-IP-30-50](https://github.com/Dlloydev/sTune/wiki/Step-Test-IP-30-50) where I've set `settleTimeSec = 600;  // 10 minutes`. If, for example your time constant is 5min, you may need to set `settleTimeSec` to something like 900-1200 (15-20minutes).

In the testing I've done, I've found that absolute steady state isn't really necessary, so there's room to relax this tolerance ... it'll require revising the internal dead time algorithm. I should be able to improve this for the next release.

robcazzaro, 2022-01-07:
Thanks for the additional info. Just one clarification, please. What variable do you refer to when you say "If, for example your time constant is 5min"?

I'm going to upload 4 separate runs shortly, where I used settleTimeSec=300 and testTimeSec=600 with samples=600, too

Dlloydev, 2022-01-07:
> What variable do you refer to when you say "If, for example your time constant is 5min"?

Just referring to the process time constant  `Tau Sec` in the print results.
You can set `testTimeSec` to something close to this. Note that `testTimeSec` acts a timeout and both `samples` and  `testTimeSec` determine the sample period.

Dlloydev, 2022-01-07:
Just to note that the updated version I pushed to the repo earlier today has tested OK with my TCLab hardware with only 300 seconds settling time (the hardware has a Tau of 175 seconds), so this was only about 1.7 Tau settling time .

closed 2022-01-08 by Dlloydev