Dlloydev/sTune #2: Reset() doesn't properly re-initialize internal variables

robcazzaro, 2022-01-06:
I'm building a hot plate for SMD soldering with a k-type thermocouple and a plate heater controlled by a SSR. I use a relay-like control for the SSR, with 2 sec window and the SSR on for a fraction of that time. I built an Arduino program to auto tune my physical setup. Here's the simplified version

```
//user settings. PWM cycle is 2 seconds, 0-100% is 0-2000
const uint32_t testTimeSec = 600;
float outputStart = 200;     // 10% start
float outputStep = 400;   // 20% step
const uint16_t samples = 600;
const uint32_t settleTimeSec = 300; //5 minutes to settle

float Input = 0, Output = 0, Setpoint = 30, Kp = 0, Ki = 0, Kd = 0;

void setup()
{
sTune tuner = sTune(&Input, &Output, tuner.zieglerNicholsPID, tuner.directIP, tuner.printALL);
  tuner.Configure(outputStart, outputStep, testTimeSec, settleTimeSec, samples);
  reflowStatus = REFLOW_STATUS_ON;
  Output = outputStart;
  /*while (true)
  {
    lcd.setCursor(0, 1);
    lcd.print(analogRead(A0));
    lcd.print("    ");
  }*/
}

void loop()
{
  // Current time
  unsigned long now;

  switch (tuner.Run())
  { // active while sTune is testing (non-blocking)
  case tuner.inOut:
    // Time to read thermocouple?
    if (millis() > nextRead)
    {
      // Read thermocouple next sampling period
      nextRead += SENSOR_SAMPLING_TIME;
      // Read current temperature
      Input = thermocouple.readCelsius();
    }
    break;

  case tuner.tunings:                                         // active just once when sTune is done
    tuner.SetAutoTunings(&Kp, &Ki, &Kd);                      // sketch variables are updated by sTune
    outputStart += 200;
    outputStep += 200;
    if (outputStep > 2001)
    {
      digitalWrite(ssrPin, LOW);
      Serial.println("end");
      while (true) {}
    }
    Output = outputStart;

    tuner.Reset();
    tuner.Configure(outputStart, outputStep, testTimeSec, settleTimeSec, samples); // restart autotune
    break;
  }

  // SSR control 
    now = millis();
    if ((now - windowStartTime) > windowSize)
    {
      // Time to shift the Relay Window
      windowStartTime += windowSize;
    }
    if (Output > (now - windowStartTime))
      digitalWrite(ssrPin, HIGH);
    else
      digitalWrite(ssrPin, LOW);
  }
}
```

As it's now. the loop runs once, and when I try to reset the autotuner to create a step response for different PWM value, it exits immediately without a settling period nor a full analysis. 

In order to make it work for me, I need to change the Reset function into 

```
void sTune::Reset() {
  _Ku = 0.0f;
  _Tu = 0.0f;
  _td = 0.0f;
  _kp = 0.0f;
  _ki = 0.0f;
  _kd = 0.0f;
  _tunerStatus = inOut;
  usPrev = micros();
  settlePrev = micros();
  pvTangentPrev = 0;
  dtCount = 0;
  ipCount = 0;
  plotCount = 0;
  sampleCount = 0;
  pvPkCount = 0; 
}
```
I'm not sure if all of those are needed, but there are a lot of internal variables that are initialized to 0 only when declared, so the second time sTune is invoked, those have the previous values. I didn't want to fully unravel all your code to see which ones really needed to be changed, so I simply reset to 0 (or the right values) all the internal variables that are declared = 0 in your code (lazy, I know :)

Also, your example code only works if the Arduino is restarted and sTune is called before a long enough time has elapsed, since it assumes that usPrev and settlePrev are close enough to 0 when sTune is called. By adding   usPrev = micros(); and   settlePrev = micros(); in the Reset() function, it works even as part of a cycle like in my case. Otherwise, if those are not initialized properly on reset(), this line of code simply skips the settling time  if (settleElapsed >= _settlePeriodUs) {           // if settling period has expired

Dlloydev, 2022-01-07:
Running multiple tests shouldn't be too difficult to implement ... I'll make this high priority for the next release (soon). Thanks for the simplified code and comments (very helpful).

Dlloydev, 2022-01-08:
resolved in sTune 1.1.0

closed 2022-01-08 by Dlloydev