Dlloydev/sTune #4: Relay Heater sTune

Kotik82, 2022-01-07:
Hello and thank u for both the QuickPID and sTune, they are both great tools to have.

I am having an issue with trying to auto tune my Silicon Heater, i am using a mosfet to control a 24v 200watt heater and use the mosfet in pseudo PWM mode (as a ON/OFF relay) to do so with 1 second intervals.

My issue is that i got no idea how to proceed with sTune since it supports PWM values of 0 to 255 and my heater will not turn on with values below 129 and will turn fully on with values above 130. 

Running the tune with the following settings:

const uint32_t testTimeSec = 300;
const float outputStart = 0;
const float outputStep = 255;
const uint16_t samples = 500;
const uint32_t settleTimeSec = 30;

Problem is heater is ON for a long period of time and i am forced to hard stop the tune cause otherwise it will burn out.

Any possible solutions?

Thank u in advance for ur hard work.

Kotik82, 2022-01-07:
You can close the issue, i went ahead with full PWM instead of relay mode, everything is working fine.

Would u please suggest directIP vs direct5T for a heater during tuning? also which Tuning Rule is better suited for my case, (yogurt maker with stable temp of around 45 degrees Celsius that should ideally never overshoot and which PWM step should i aim during tune, i went with:

const uint32_t testTimeSec = 300;
const float outputStart = 0;
const float outputStep = 30;
const uint16_t samples = 500;
const uint32_t settleTimeSec = 300;

Not sure if these setting are suited for my environment.

Dlloydev, 2022-01-07:
Hello an thanks for posting your setup and details.
> Would u please suggest directIP vs direct5T for a heater during tuning?

You can use either one, but directIP will complete its test much faster.

>  also which Tuning Rule is better suited for my case, (yogurt maker with stable temp of around 45 degrees Celsius that should ideally never overshoot

Progressively more conservative rules are `tyreusLuybenPID`, `someOvershootPID` and `noOvershootPID`

> and which PWM step should i aim during tune

Your setup and step size of 30 looks good ... glad everything is working fine!

Kotik82, 2022-01-08:
Hello and thank u for the suggestions, am i correct in my understanding that:
const uint32_t testTimeSec = 400;
const uint16_t samples = 400;

will force sTune to sample once every second?

The reason i am asking is cause SetSampleTimeUs from QuickPID is very useful in my case (ds18b20 with 200ms delay on a refresh of temperature) to set the correct PID timer. And since sTune doesnt provide it in the same form i am guess that u can manipulate it over testTimeSec & samples.

Please correct me if i am wrong, i want to make sure i get at least 3 new temp readings per sTune cycle.
Thank u in advance.

Dlloydev, 2022-01-08:
Yes, that works out to once every second. If you need a longer sample time, just increase testTimeSec and/or lower samples. For example, with testTimeSec = 600  and samples = 200, the sample period is 3 seconds.
A good range for samples is 200-500.

Kotik82, 2022-01-08:
Thank you very much.

I think i will eventually go for a thermistor instead of DS18B20, since it is slow and i noticed that when i lower its resolution to 9 (so 94ms delay between new temp reading) and set QuickPID to myPID.SetSampleTimeUs(300000); i am getting less overshooting so i guess if i had a sensor that could read faster (ie thermistor) i would have even better results.

Kotik82, 2022-01-08:
I remember reading somewhere that ideally we want our sensor to be able to push 3 temp readings per sample, is this correct? No idea why this would be needed since we do our calculations based on 1 sample and as a result would need 1 correct sample right before doing our calculation. As an example DS18B20 can push a fresh reading every 94ms and if we set myPID.SetSampleTimeUs(100000); we should be getting a fresh reading for ever calculation. Unless we can have some misalignment between the request which will end up in a not synced temp reading (ie always getting previous reading instead of fresh).

This is basically what i am using to read the sensor:
double temperature;
unsigned long lastTempUpdate;
bool updateTemperature() {
  if ((millis() - lastTempUpdate) > TEMP_READ_DELAY) {
    temperature = sensors.getTempCByIndex(0); //get temp reading
    lastTempUpdate = millis();
    sensors.requestTemperatures(); //request reading for next time
    return true;
  }

I know this might be off topic but it is still interesting to know.

Dlloydev, 2022-01-08:
Yes, I agree that you only need to get at minimum 1 new reading per sample for sTune.
It's more important to get at least 1 new reading per PID compute cycle (pid sample rate).

There's an interesting comment near the end of this link [here](https://electronics.stackexchange.com/questions/538459/relationship-between-latency-and-sampling-rate-of-a-digital-pid-controller).

> _For this reason, it is common to see the sampling rate >=10x the f(max) rather than 2.5 to 3x for high SNR signals relative to control systems..._

Example: process time constant Tau = 400 seconds
If we use f(max) = 1/Tau = 1/400  = 0.0025Hz
For sampling rate >= 10 x f(max) we get 0.025Hz or higher
This is the same as a sampling period of 40 seconds or less.

Kotik82, 2022-01-09:
Nice link, it was a good read, and this one is even more so relevant when it comes to PID controlled heating elements since they always have lag when it comes to PID response and by forcing higher sampling rate u minimize the aliasing phenomenon.

Trying to force a higher sampling rate is what i was testing all day yesterday, the BIG issue of DS12B20 is that when u try to push higher sampling rates u loose resolution :( grrrrrrr

In its fastest polling rate (ie 94ms) its resolution is limited to +-0.5 degrees which as u can imagine will again introduce aliasing, cause while u poll fresh data from the sensor every 94ms and have a sampling rate of ie 100ms u are guaranteed to get a fresh reading BUT it "might" be within the same +-0.5 degrees resolution (ie 40.51 & 40.74 will both report 40.50 for both readings). It ends up being a pick your poison situation and i don't think DS12B20 is suited for tight temp control PID systems exactly due to its poor resolution on fast sampling rates.

What is left for me to test today is setting the sensor to its max resolution and go for a sampling rate of 1000ms or higher, until now the sweet spot is a resolution of +-0.25 with a sampling rate of 600ms.

Thermistor is better suited for my project and i will go for it sooner or later, i am just too scared of getting accurate readings out of them hence why i went for a digital sensor. I guess i will have to see start reading how to get accurate readings from a thermistor and also which thermistor is good for my case (btw if u have any suggestions let me know).

Both of your libraries are extremely useful to me and fun to play with.

Dlloydev, 2022-01-09:
> What is left for me to test today is setting the sensor to its max resolution and go for a sampling rate of 1000ms or higher, until now the sweet spot is a resolution of +-0.25 with a sampling rate of 600ms.

By tomorrow, I should have a new version of sTune ready. Presently, the tuning constants are being derived incorrectly and aren't ideal for slower systems and when using relay output control.

You should notice a significant improvement  ... especially when using longer sampling intervals and (hopefully) relay output control. I say "hopefully" because I don't have a hardware setup to test this (yet).

The new sTune version will require entering the span for input and output.

**Examples:**
If your input works with temp readings of 20C min and 150C max, then `inputSpan = 130;`
If the output uses 8-bit PWM and your using its full range, then `outputSpan = 255;`
If the output is digital relay controlled by millis() with window period of 2000ms, then `outputSpan = 2000;`

Kotik82, 2022-01-09:
Yea cause i noticed that when i use the suggested gains after sTune i am always sitting below the setpoint no matter how long i wait BUT if i import the values from printPIDTUNER into pidtiner.com i am getting much better results, i tried with noOvershootPID, tyreusLuybenPID (both cannot reach setpoint) and zieglerNicholsPID (this one overshoots by 2 degrees and jumps between setpoint/+2 degreees.

But i am not using a relay what i do use is a sampling rate of 1000ms.

Hope to see improvements.

Dlloydev, 2022-01-11:
sTune 2.0.0 is published ... new open loop tuning methods and updated examples.

Kotik82, 2022-01-11:
Great news, i am reading right now, got a question already, everything seems to make sense besides inputSpan, how do we ideally determine that value? Do we just check our sensor range or do we set values based on our real environment.
 
To clarify my sensor has a range of -55 to 125 degrees so if we go by the raw sensor specs then inputSpan = 180 but i know for a fact my readings will never drop below 0.

Dlloydev, 2022-01-11:
That's good to hear!

> _To clarify my sensor has a range of -55 to 125 degrees so if we go by  the raw sensor specs then inputSpan = 180 but i know for a fact my  readings will never drop below 0._

In this case, `inputSpan` could be 125. This parameter is used in the reported process gain and only in the formulas for `kp` in the `NoOvershoot_PID` and `NoOvershoot_PI` tuning methods. Using 125 would increase the`kp` result and cause the tuning for this method to be a bit more aggressive.

Check out the new [Get_All_Tunings](https://github.com/Dlloydev/sTune/blob/main/examples/Get_All_Tunings/Get_All_Tunings.ino) example where you get results for all 10 tuning methods in just 1 quick test run. Here's the print results from my TCLab hardware:
[Get_All_Tunings.txt](https://github.com/Dlloydev/sTune/files/7846756/Get_All_Tunings.txt)

Kotik82, 2022-01-11:
Controller Action: direct5T
 Tuning Method:     NoOvershoot_PID

 Output Start:      0.00
 Output Step:       20.00
 Sample Sec:        1.00000

 Pv Start:          28.7500
 Pv Max:            45.3013
 Pv Diff:           16.5513

 Process Gain:      1.1724
 Dead Time Sec:     41.39423
 Tau Sec:           253.61416

 Tau/Dead Time:     6.1 (easy to control)
 Tau/Sample Period: 253.6 (good sample rate)

 Kp: 3.1356
 Ki: 0.0237
 Kd: 0.2898

 ZN_PID
  Kp: 2.72
  Ki: 0.07
  Kd: 0.29

 ZN_Half_PID
  Kp: 1.36
  Ki: 0.04
  Kd: 0.14

 Damped_PID
  Kp: 1.49
  Ki: 0.09
  Kd: 0.21

 NoOvershoot_PID
  Kp: 3.14
  Ki: 0.02
  Kd: 0.29

 CohenCoon_PID
  Kp: 3.71
  Ki: 0.06
  Kd: 0.41

 ZN_PI
  Kp: 2.04
  Ki: 0.04
  Kd: 0.00

 ZN_Half_PI
  Kp: 1.02
  Ki: 0.02
  Kd: 0.00

 Damped_PI
  Kp: 1.12
  Ki: 0.06
  Kd: 0.00

 NoOvershoot_PI
  Kp: 1.83
  Ki: 0.02
  Kd: 0.00

 CohenCoon_PI
  Kp: 2.47
  Ki: 0.06
  Kd: 0.00

This is with inputSpan 180, will try later on with 125 as suggested and report back, i noticed that NoOvershoot_PID is working great while ZN_PID just cannot get to Setpoint and the P value of it suggests that it is less aggressive than NoOvershoot, is this working as intended?

Dlloydev, 2022-01-12:
**IMPORTANT**: Need to reset Output to 0 prior to switching PID to automatic, otherwise this will create an initial excessive overshoot.

```c++
    case tuner.tunings:                                          // active just once when sTune is done
      tuner.GetAutoTunings(&Kp, &Ki, &Kd);                       // sketch variables updated by sTune
      myPID.SetSampleTimeUs((testTimeSec * 1000000) / samples);  // PID sample rate (same as sTune)
      Output = 0;                                                // required prior to automatic mode
      myPID.SetMode(myPID.Control::automatic);                   // the PID is turned on (automatic)
      myPID.SetTunings(Kp, Ki, Kd);                              // update PID with the new tunings
      break;

```
An update to follow in a few days with updated examples, plotter function, tests, documentation, etc.

Dlloydev, 2022-01-16:
Just released a new version that should work with relay, MOSFET or transistor PID control.

Kotik82, 2022-01-17:
Good job, everything is working fine on my end, i am using hardware PWM so i didnt test the new relay features, but i did use sTune to tune PID and import values into QuickPID.

All is great, just a question regarding a new option "clearPidOutput" should i set it to 0 if u wanna do a fresh sTune run ONLY and then manually copy the values? This new option is a bit confusing to me.

Dlloydev, 2022-01-17:
Glad its working great,

Yeah, you don't need to use the `clearPidOutput` option. If you comment it out, then the PID will start with the output value that sTune completed with (the `outputStep` value).

Changing the output will have no effect on sTune because it uses its own `outputStart` and `outputStep` values. However, when doing an autotune run (**sTune ➩ QuickPID**), there's an opportunity to change the starting output value that QuickPID uses (if needed).

Thanks for your feedback.   -dlloyd

Dlloydev, 2022-01-17:
I'll close this for now, but feel free to re-open or start a new issue or discussion if needed.

closed 2022-01-17 by Dlloydev