Dlloydev/sTune #5: Relay heater example

Hunter6400, 2022-01-11:
Hello and thank you for the great library and project. 
Would it be possible to add example on how to tune relay output pid. I have mosfet switching heater. Or would you suggest some better way then using the mosfet as relay. 
I know this thread has been opened before, but I quite don't understand how is was solved.
Thank you very much.
Also this is my first time with github so sory if this is not the proper way to do this.

Dlloydev, 2022-01-11:
Hello,
Yes, that's a good idea. While developing this library I've been using PWM, but will do some testing using a digital ON/OFF output for temperature control. Could take a week or two because I'm quite far behind with other things, but I'll make this a high priority for the next example.

Hunter6400, 2022-01-14:
Hello can I have a simple question? How do you implement pwm control? Do you have to have regulator which works on different voltage levels and it is controlled by the analog output? Or do you use pwm as a fast switching signal for mosfet? Thank you so much.

Hunter6400, 2022-01-14:
I am using ESP 32 wroom DEVKITc module. Back from some internet learning. I understand that the output that the library is sending is in fact 1 and 0 with different duty cycle? I hope my understanding is right, but could I theoretically use this to switch mosfet on and off the mosfet has enough switching speed I am sure of that. So I don't have to deal with using the mosfet as a relay, which is harder because relay are slow af. Will this also work on esp32 it should, but I had problems with other libraries already? Thank you and I must say this is by far the best pid implementation in arduino. Have a wonderful day.

Dlloydev, 2022-01-14:
Thanks,

I happen to have the same ESP module on hand. Should have no trouble controlling a mosfet, providing it's logic level type with VGS rating of 2.7V or less and has a working current and voltage rating suitable for your heater and its power source.

The library can use hardware PWM (using the analogWrite function) or it can just switch an output on and off (using the digitalWrite function) over a longer period or "window" of time (software PWM). Either way should work with a mosfet controlling a heater. I'll have a working example using software PWM in the next day or two.

Dlloydev, 2022-01-16:
Just released a new version that should work with relay, MOSFET or transistor PID control.

Dlloydev, 2022-01-22:
There's now updated examples provided. Please re-open if you have any questions
Regards, -dlloyd

closed 2022-01-22 by Dlloydev

lpearl, 2022-07-25:
Hey, @Dlloydev this is fantastic work! I'm trying to heat a larger thermal mass but it takes too long to heat up since the SSR is pulsing on and off.

Any ideas?

Dlloydev, 2022-07-25:
Thanks,
Could you open a discussion on this?
Also, If you could describe a bit more about your system and show your example code it would be really helpful.
Thanks again,
David