Dlloydev/sTune #10: PTC Heater Requires Using `direct5T` Test

Dlloydev, 2022-02-05:
If running sTune and using a PTC heater, the much quicker `directIP` test appears to produce good results for everything except for its ability to accurately determine PV max ... this also results in Kp being inaccurate (too high).

Presently I'm using a thermal calculation to get PV Max which doesn't account for the positive temperature coefficient of the PTC, so I'll need to investigate. I may need to add a new `IP` test mode for PTC because the `IP` test is just too fast (240 sec vs 1122 sec) and too close to being ready to ignore. See comparison in image below:

[image]

geleos27, 2022-02-05:
Will this also affect on other type of heaters with even bigger dead time and inertia?

Dlloydev, 2022-02-05:
Thermal type heaters should should be OK with using the `directIP` test.

Note that the software does a check to see how controllable the process is (Tau / Dead Time). If you suspect any issue with the `directIP` results, then try running `direct5T` to compare results.

Dlloydev, 2022-02-13:
Resolved in sTune 2.4.0

closed 2022-02-13 by Dlloydev