Dlloydev/sTune #13: optimumOutput from softPwm looks incorrect

bakingbaking, 2022-06-05:
SoftPwm for SSR (delay == 0) looks like it is trying to deal with the extra time due to turn off at a zero crossing when computing optimumOutput.
Since we don't know where our window started in a cycle we will get anywhere from 0 to 0.5 extra cycle of output.
Unless we know where we start, I think the best we could do is make it so we get +/- 0.25 of a cycle by subtracting 0.25 of a cycle.
For 60 Hz that would be 4 mS or 4.1666667 mS, for 50 Hz that would be 5 mS.
I think this is true whether we are above or below the set point.

So

  if (input > setpoint) reachedSetpoint = true;
  if (reachedSetpoint && !debounce && setpoint > 0 && input > setpoint) optimumOutput = output - 8;
  else if (reachedSetpoint && !debounce && setpoint > 0 && input < setpoint) optimumOutput = output + 8;
  else  optimumOutput = output;
  if (optimumOutput < 0) optimumOutput = 0;

would become:

  if (input > setpoint) reachedSetpoint = true;
  if (reachedSetpoint && !debounce && setpoint > 0 && input > setpoint) optimumOutput = output - 4;
  else  optimumOutput = output;
  if (optimumOutput < 0) optimumOutput = 0;

In reality, if the PID has an "I" component it will correct for the offset so this all may be a moot point.

bakingbaking, 2022-06-05:
Sorry, I left in the setpoint test.  The code should be:

if (input > setpoint) reachedSetpoint = true;
if (reachedSetpoint && !debounce && setpoint > 0 ) optimumOutput = output - 4;
else optimumOutput = output;
if (optimumOutput < 0) optimumOutput = 0;

bakingbaking, 2022-06-05:
Maybe we don't even need the setpoint test:

if (!debounce && setpoint > 0 ) optimumOutput = output - 4;
else optimumOutput = output;
if (optimumOutput < 0) optimumOutput = 0;

bakingbaking, 2022-06-05:
I wanted to add, thank you for both of your terrific PID packages!

Dlloydev, 2022-06-06:
Thank you for your interest in this and your comments.

The control resolution for PID or sTune is 1 ms, but when using an SSR each half cycle represents 8.33ms for 60Hz AC, or 10ms for 50Hz AC. When controlling a load at low power (say we only need 10 cycles ON in each second to maintain the power at setpoint), then each half cycle would represent 5% of the required power. In this case, there would be a dead-band of 5% power before the change in output has any effect. This issue causes drifting of the input about the setpoint resulting in poor control resolution and additional overshoot. This issue becomes less significant when controlling at higher load levels as we get closer to the upper control limit.

SoftPWM tackles this issue by dropping out or adding in almost 1 AC cycle of control value to the output, but only when the input crosses the setpoint. This oscillatory control action about the setpoint improves response and provides tighter control.

_Example plot showing optimum output SSR control, which has the effect of reducing overshoot and maintaining tighter regulation at the setpoint:_

[image]

Here, you can see the initial half cycle dropout in the output just as the input crosses above the setpoint. Then, the output gradually, but not completely, enters a half cycle oscillatory action. The hardware used was an SSR controlling a 140W, 220℃ PTC heater with temperature setpoint at only 80℃.

bakingbaking, 2022-06-06:
Thanks for the quick response.
What is the time scale of the graph?
Also, please what outputSpan and tuning rule was used for the example?

Dlloydev, 2022-06-07:
I had the outputSpan set to 1000 and was plotting every 3rd sample, so each 100 on the scale is 300 sec (5 min).

bakingbaking, 2022-06-07:
Thank you so much!
I just realized the plot is from one of the sTune_QuickPID_Less_Overshoot examples.
This was really helpful.

closed 2022-06-07 by bakingbaking