Dlloydev/sTune #19: "outputStep" is not working correct

Farhankhosravi, 2022-08-02:
Hi
I recently used sTune library for a MAX6675 Furnace... it was hard to tune it manually as temperature changes very slow.
So I decided to use sTune, and tried to edit configuration numbers as best as i can... but, my problem is on "outputStep"...
this parameter is so weird. its default value is 50, and system will start with 50/1000 of heater output. this is so weak to even make my furance a little warm! so i increased it to 700, and test started and finished (but unfortunately every time it gaves a different p-i-d numbers.. anyway).. so:
it is expected that sTune find best "outputStep" automatically; and do no need to be changed by user... 
and an **IMPORTANT** notice, if your setpoint is below current temprature (by mistake, running sTune in midde of work, etc) it **wont** notice that setpoint is below input and starts with adjusted "outputStep" value, makes your furnace over heated and makes burnings/etc..
please take a look on "outputStep". i think it needs a revision.
Thanks

Dlloydev, 2022-08-02:
Hello and thank you for your feedback.

The examples using the MAX6675 were for a PTC type heater. I plan on doing a revision later this year that includes examples using a regular thermal heater and provide improved documentation and easier to use interface.

Yes, sTune doesn't use setpoint while it determines the PID gains, so its important to set the `tempLimit` value prior to running the test.

To get more consistent PID numbers for a given `outputStep` (stepped power level), then its important that enough time has elapsed while resting at the `outputStart` value (power level) for the temperature to be stable before starting the test.

If needed, you can enforce a cool-down period between tests for the input temperature to stabilize. For example, use `settleTimeSec = 1800;` for 30 minutes cool-down time.

InnuendoPi, 2022-08-30:
Hey,

> I plan on doing a revision later this year that includes examples using a regular thermal heater and provide improved documentation and easier to use interface.

that would be helpful! I also stuck on testTimeSec, outputspan and outputstep.
My environment: ESP8266, QuickPID, kettle (outputspan: 0..100% powerlevel, inductioncooker with seperat control functions), DS18B20 (inputspan 0..100°C). hints and tips are wlecome.

HuegelMensch, 2024-06-09:
did u found something ? its pretty hard from scratch and sTune wont work, i tried for days.