Dlloydev/sTune #28: output always nan

zekageri, 2023-01-27:
First of all, great library and thank you for the effort.

I'm using an ESP32 Arduino.

I have an implementation where a user can create any number of `thermSystems` which is represented as a class for me.
In every `thermSystem` there is it's own `set` and `measured` temperature coming from different sensors.
For every `thermSystem` there is only one `boiler` with it's own input which controlls if it has to start to heat or not.
For me, the esp can not directly set a pin to high or low, so i modified your library a little bit in the following ways:

**sTune.h**
```cpp
float softDigit(boolean &heatControllBit, float input, float output, float setpoint, uint32_t windowSize, uint8_t debounce);
```
**sTune.cpp**

```cpp
float sTune::softDigit(boolean &heatControllBit, float input, float output, float setpoint, uint32_t windowSize, uint8_t debounce) {

  // software PWM timer
  uint32_t msNow = millis();
  static uint32_t  windowStartTime, nextSwitchTime;
  if (msNow - windowStartTime >= windowSize) {
    windowStartTime = msNow;
  }
  // SSR optimum AC half-cycle controller
  static float optimumOutput;
  static bool reachedSetpoint;

  if (input > setpoint) reachedSetpoint = true;
  if (reachedSetpoint && !debounce && setpoint > 0 && input > setpoint) optimumOutput = output - 8;
  else if (reachedSetpoint && !debounce && setpoint > 0 && input < setpoint) optimumOutput = output + 8;
  else  optimumOutput = output;
  if (optimumOutput < 0) optimumOutput = 0;

  // PWM relay output
  static bool relayStatus;
  if (!relayStatus && optimumOutput > (msNow - windowStartTime)) {
    if (msNow > nextSwitchTime) {
      nextSwitchTime = msNow + debounce;
      relayStatus = true;
      heatControllBit = true;
    }
  } else if (relayStatus && optimumOutput < (msNow - windowStartTime)) {
    if (msNow > nextSwitchTime) {
      nextSwitchTime = msNow + debounce;
      relayStatus = false;
      heatControllBit = false;
    }
  }
  return optimumOutput;
}
```

As you can see i have added a modification to the softpwm function to be able to set a `boolean`  variable by `value` instead of a `digitalWrite`. This is to ensure that no pin is changed in my setup and controlls only at software level.

So with that in mind, i have the following controll functions for the auto tune PID.

```cpp
/* PID DETAILS */
const uint8_t inputPin = 0;

uint32_t settleTimeSec = 10;
uint32_t testTimeSec = 1000;
const uint16_t samples = 500;
const float inputSpan = 150;
const float outputSpan = 1000;
float outputStart = 0;
float outputStep = 300;
float tempLimit = 350;

// variables
float Input, Output, Setpoint = 50, Kp, Ki, Kd;
sTune *tuner;
QuickPID *myPID;

void pidInit();
void pidLoop();
/* PID DETAILS */

/*
*     This function is called at startup of the thermSystem
*/
void thermSystem::pidInit() {
    tuner = new sTune(&Input, &Output, tuner->ZN_PID, tuner->directIP, tuner->printOFF);
    myPID = new QuickPID(&Input, &Output, &Setpoint);
    tuner->Configure(inputSpan, outputSpan, outputStart, outputStep, testTimeSec, settleTimeSec, samples);
    tuner->SetEmergencyStop(tempLimit);
}

/*
*     This function is called in every iteration of the loop()
*/
void thermSystem::pidLoop() {
    tuner->softDigit(shouldStartHeat, Input, Output, 0, outputSpan, 1);

    switch (tuner->Run()) {
        case tuner->sample:  // active once per sample during test
            Input = currentTemp;
            //Input = analogRead(inputPin) * 0.322265625 - 50.0;  // get degC (using 3.3v AREF)
            tuner->plotter(Input, Output * 0.1, Setpoint, 1, 3);
            break;

        case tuner->tunings:                       // active just once when sTune is done
            tuner->GetAutoTunings(&Kp, &Ki, &Kd);  // sketch variables updated by sTune
            myPID->SetOutputLimits(0, outputSpan);
            myPID->SetSampleTimeUs(outputSpan * 1000 - 1);
            myPID->SetMode(myPID->Control::automatic);  // the PID is turned on
            myPID->SetProportionalMode(myPID->pMode::pOnMeas);
            myPID->SetAntiWindupMode(myPID->iAwMode::iAwClamp);
            myPID->SetTunings(Kp, Ki, Kd);  // update PID with the new tunings
            break;

        case tuner->runPid: // active once per sample after tunings
            //Input = analogRead(inputPin) * 0.322265625 - 50.0;  // get degC (using 3.3v AREF)
            Input = currentTemp;
            myPID->Compute();
            tuner->plotter(Input, Output, Setpoint, 0.1f, 3);
            break;
    }
}
```

I have the following output on the serial monitor:

```cpp
Setpoint:150.00, Input:197.00, Output:nan,
Setpoint:225.00, Input:0.00, Output:nan,
Setpoint:210.00, Input:0.00, Output:nan,
Setpoint:210.00, Input:0.00, Output:nan,
Setpoint:230.00, Input:0.00, Output:nan,
Setpoint:150.00, Input:198.00, Output:nan,
Setpoint:225.00, Input:0.00, Output:nan,
Setpoint:210.00, Input:0.00, Output:nan,
Setpoint:210.00, Input:0.00, Output:nan,
Setpoint:230.00, Input:0.00, Output:nan,
Setpoint:150.00, Input:198.00, Output:nan,
Setpoint:225.00, Input:0.00, Output:nan,
Setpoint:210.00, Input:0.00, Output:nan,
Setpoint:210.00, Input:0.00, Output:nan,
Setpoint:230.00, Input:0.00, Output:nan,
Setpoint:150.00, Input:199.00, Output:nan,
Setpoint:225.00, Input:0.00, Output:nan,
Setpoint:210.00, Input:0.00, Output:nan,
Setpoint:210.00, Input:0.00, Output:nan,
Setpoint:230.00, Input:0.00, Output:nan,
Setpoint:150.00, Input:199.00, Output:nan,
Setpoint:225.00, Input:0.00, Output:nan,
Setpoint:210.00, Input:0.00, Output:nan,
Setpoint:210.00, Input:0.00, Output:nan,
Setpoint:230.00, Input:0.00, Output:nan,
Setpoint:150.00, Input:199.00, Output:nan,
Setpoint:225.00, Input:0.00, Output:nan,
```

And my heating can not turn off. 
It is always on but the current set temperature is 15.5 celsius which is represented as 155 decimal and the current temperature is 20 celsius which is 200 decimal.

current temperature value is coming in Serial modbus communication which sample rate is roughly equal to the loop iteration in my setup.

**What could be the problem?**

Dlloydev, 2023-01-28:
Thank you for your detailed information on your system and tests.

I've created an [example in Wokwi](https://wokwi.com/projects/355040432629843969) perhaps you could look at.
Your function seems to be linking OK to the `boolean shouldStartHeat;` variable and the Output isn't showing nan.

Still, on your system, you may need to use a global variable in the in the class ... `*_shouldStartHeat` in a way similar to how `*_input` and `*_output` is used in the header file and .cpp file.

The valve open time which varies 60 to 600 seconds I think would be the deadtime. This would be 6 to 60% of your test time  `uint32_t testTimeSec = 1000;`.(use a testTimeSec value that is an estimate of the time constant of your system). Here's a [chart](https://blog.opticontrols.com/wp-content/uploads/2011/06/td-versus-tau.png) that shows how difficult a system will be to control depending on the ratio of deadtime to tau (time constant). You might have to use PI control rather than PID to get more range of controllability.

zekageri, 2023-01-28:
Thank you for the response.
The shouldStartHeat is indeed global in the class, but it isn't a pointer. I will try this and will take a look at the example. Thank you very much. As of the valve time. It is an input which a user can control depending on he's phisycal valve type. So I think I will make the dead time equal to this time.

Will test your suggestions and come back to you!

zekageri, 2023-01-29:
In the example you created at wokwi, the led is just blinking and does not stay on.

zekageri, 2023-01-29:
Do you have any idea why is my input sometimes 0 when plotting to Serial? I have consistent readings

```
Setpoint:230.00, Input:0.00, Output:5.00,
Setpoint:210.00, Input:0.00, Output:5.00,
Setpoint:170.00, Input:176.00, Output:5.00,
Setpoint:225.00, Input:0.00, Output:5.00,
Setpoint:210.00, Input:0.00, Output:5.00,
Setpoint:230.00, Input:0.00, Output:5.00,
Setpoint:210.00, Input:0.00, Output:5.00,
Setpoint:170.00, Input:178.00, Output:5.00,
```

zekageri, 2023-01-30:
Really confused about why the `Setpoint` and `Input` is changing while these should be constants??

I have enabled every serial output and here is the results. Right now the set temperature is 200, the measured is 191 and it does not set the boolean value to true at all. Sometimes it does set it but it does not turn it off ever.

```cpp
Setpoint:225.00, Input:0.00, Output:0.00,
 sec: 140.1491  out: 100.00  pv: 0.000  tan: -111.500 →
 sec: 145.1540  out: 100.00  pv: 223.000  tan: 111.500 →
Setpoint:230.00, Input:223.00, Output:10.00,
 sec: 145.1529  out: 100.00  pv: 0.000  tan: -111.500 →
Setpoint:210.00, Input:0.00, Output:10.00,
 sec: 150.1546  out: 100.00  pv: 223.000  tan: 111.500 →
Setpoint:210.00, Input:212.00, Output:nan,
 sec: 150.1611  out: 100.00  pv: 0.000  tan: -111.500 →

 Controller Action: directIP

 Output Start:      0.00
 Output Step:       100.00
 Sample Sec:        5.0000

 Pv Start:          223.000
 Pv Max:            663.026
 Pv Diff:           440.026

 Process Gain:      25.144
 Dead Time Sec:     0.000
 Tau Sec:           188.705

 Tau/Dead Time:     99.9 (easy to control)
 Tau/Sample Period: 37.7 (good sample rate)

 Tuning Method: ZN_PID
  Kp: inf
  Ki: inf  Ti: nan
  Kd: inf  Td: nan

Updating with new tunings: Kp: inf, Ki: inf, Kd: inf
Setpoint:200.00, Input:191.00, Output:0.00,
Setpoint:225.00, Input:0.00, Output:0.00,
 sec: 155.1621  out: 100.00  pv: 0.000  tan: -111.500 →
 sec: 160.1693  out: 100.00  pv: 0.000  tan: 92.917 ↗
Setpoint:210.00, Input:0.00, Output:10.00,
Setpoint:230.00, Input:223.00, Output:nan,
Setpoint:210.00, Input:212.00, Output:nan,
 sec: 165.1763  out: 100.00  pv: 0.000  tan: -130.083 ↘
Setpoint:200.00, Input:191.00, Output:0.00,
Setpoint:225.00, Input:0.00, Output:0.00,
 sec: 170.1812  out: 100.00  pv: 0.000  tan: 74.333 ↗
 sec: 175.1822  out: 100.00  pv: 0.000  tan: -148.667 ↘
Setpoint:210.00, Input:0.00, Output:10.00,
Setpoint:230.00, Input:223.00, Output:nan,
Setpoint:210.00, Input:212.00, Output:nan,
 sec: 180.1889  out: 100.00  pv: 0.000  tan: 55.750 ↗
Setpoint:200.00, Input:191.00, Output:0.00,
Setpoint:225.00, Input:0.00, Output:0.00,
```

FelixWT, 2023-02-06:
> In the example you created at wokwi, the led is just blinking and does not stay on.
> 
> Shouldnt it stay on to let the heating heat?

It's blinking because of these two lines

```
const float outputSpan = 1000;
float outputStep = 50;
```

in words this means in 1000 millisec control window, the system will send HIGH signal to pin 22 for 50 millisec.

you might ask how about set this to 

```
const float outputSpan = 1000;
float outputStep = 1000;
```

as far as I know this might lead to the searching of the inflection point to fail , you will reach temperture limit before the search is done

FelixWT, 2023-02-06:
> Do you have any idea why is my input sometimes 0 when plotting to Serial? I have consistent readings

I'll say this might be the problem on the data collecting end of you system  but I might be wrong
Can you share your code where you assign value to `currentTemp`

zekageri, 2023-02-06:
**Thank you for your response, i appreciate it.**

the `currentTemp` is coming from a `modbus RTU slave`. It is a thermostat which measures the temperature and i just assign a buffer value from the given index to the `currentTemp` variable. It does not change if i print out my readings directly from the buffer

`currentTemp = modules[thermModule.index]->readBuffer[1];`

the output must stay HIGH since this is the controll bit of the heating system. If this bit is high, the system will start the heating process. It will start to open the valves, starts the boiler and starts the circulation pump after the valves are opened. This is a relatively long process so the controll (output) bit can not fluctuate.

Right now the PID is either left the output bit on high in all cases or does not at all.

FelixWT, 2023-02-06:
You can try get rid of the PID Library first , and print out temperature every 2 sec. Just for sure that the reading logic of your system is on point.

zekageri, 2023-02-06:
I have a webserver whichs shows the measured temperature. It is refreshed every 1 sec. I see that it is correct

FelixWT, 2023-02-07:
I guess I'll provide my code since I'm using ESP32 as well...
note that I'm not using pointers to construct class instance

**main.cpp**
```
#include <Arduino.h>
#include "ArduinoJson.h"
#include "AutoTuneEvent.h"

void setup(){
  Serial.begin(115200);
  Serial2.begin(921600);
  Serial2.setTimeout(10);
  AutoTune.begin();
}

void loop(){
  server.handleClient();
 
  if(Serial2.available()){
    String RCV = Serial2.readString();
    Display_Event.rxDataSet(RCV);
//get data from Json
    StaticJsonDocument<512> doc;
    DeserializationError error = deserializeJson(doc, RCV);
    if (error) {
      String Emsg="DeserializeJson Failed";
      EspNow_Event.send(Emsg);
      return;
    }else{
// Where I assign temperature value from UART    
      float val = doc["val"];
      AutoTune.Input = val;
    }
  }  
  AutoTune.Autotuning();
}
```

**AutoTuneEvent.cpp**
```
#include "AutoTuneEvent.h"
#include "LittleFSEvent.h"
#include "ArduinoJson.h"
#include "timerEvent.h"
#include <QuickPID.h>
#include <sTune.h>
#include <Wire.h>

AutoTuneEvent AutoTune = AutoTuneEvent();
sTune tuner = sTune(&AutoTune.Input, &AutoTune.Output, tuner.ZN_PID, tuner.directIP, tuner.printDEBUG);
QuickPID myPID(&AutoTune.Input, &AutoTune.Output, &AutoTune.Setpoint);

//Get Accurate InputSpan
bool AutoTuneEvent::settleInputSpan()
{
  if(Serial2.available()){
    String UART2 = Serial2.readString();
    StaticJsonDocument<512> JBuffer;
    DeserializationError error = deserializeJson(JBuffer, UART2);
    if (error) {
      return false;
    }else{
      float  temp= JBuffer["val"];
      Input = temp;
      inputSpan=Setpoint-temp;
      return true;
    }
  }
  return false;
[... 178 lines cut ...]
  if (input > setpoint){
    reachedSetpoint = true;
  }

  optimumOutput = output<0 ? 0:output;

  // Output
  static bool relayStatus;
  if (!relayStatus && optimumOutput > (msNow - windowStartTime)) {
    if (msNow > nextSwitchTime) {
      nextSwitchTime = msNow + debounce;
      relayStatus = true;
      snprintf(HeatUpSpan, sizeof(HeatUpSpan),"{\"HOT\":[%.3f]}",optimumOutput);
      Serial.println(HeatUpSpan);
      Serial2.write(HeatUpSpan);
    }
  } else if (relayStatus && optimumOutput < (msNow - windowStartTime)) {
    if (msNow > nextSwitchTime) {
      nextSwitchTime = msNow + debounce;
      relayStatus = false;
    }
  }
  return optimumOutput;
}
```

zekageri, 2023-02-07:
Thank you for the snippet. I will read it carefully! :D

FelixWT, 2023-02-07:
I'll recommand you don't look it to following part, the logic wasn't  establish correctly

**AutoTuneEvent::begin()**

> myPID.SetMode(myPID.Control::automatic); // the PID is turned on
> myPID.SetProportionalMode(myPID.pMode::pOnError);
> myPID.SetAntiWindupMode(myPID.iAwMode::iAwOff);

**AutoTuneEvent::Autotuning()**
> else{
>     if(NormalTemp || Input < 40){
>       if(!NormalTemp){
>         NormalTemp = true;
> // Work around logic for Weird output
>       }else if(!PrePIDinFlash){
>         esp_restart();
>       } [...]

zekageri, 2023-02-20:
Thank you for the clarification. I read trought your sketch, however i can not adapt it to my system properly.
No matter what option i update, the PID always doing a PWM on my controll variable.

I adapted my `pidLoop()` this way:

```cpp
void thermSystem::adaptInputSpan(){
    if(prevSetPoint != Setpoint){
        prevSetPoint = Setpoint;
        inputSpan = Setpoint - Input;
        tuner->Configure(
            inputSpan,
            outputSpan,
            outputStart,
            outputStep,
            testTimeSec,
            settleTimeSec,
            samples
        );
    }
}

void thermSystem::pidLoop() {
    if(!isPidEnabled || !pidInited ){return;}
    adaptInputSpan();
    tuner->softDigit(shouldStartHeatCtr, Input, Output, Setpoint, outputSpan, 1);

    if( shouldStartHeat && !shouldStartHeatCtr){
        if( millis() - lastShouldStartHeatCtr >= 50000 ){
            shouldStartHeat = false;
            lastShouldStartHeatCtr = millis();
        }
    }else if( !shouldStartHeat && shouldStartHeatCtr ){
        if( millis() - lastShouldStartHeatCtr >= 50000 ){
            shouldStartHeat = true;
            lastShouldStartHeatCtr = millis();
        }
    }

    switch (tuner->Run()) {
        case tuner->sample:  // active once per sample during test
            Input = currentTemp;
            tuner->plotter(Input, Output * 0.1, Setpoint, 1, 3);
            break;

        case tuner->tunings:    // active just once when sTune is done
            tuner->GetAutoTunings(&Kp, &Ki, &Kd);
            myPID->SetTunings(Kp, Ki, Kd);
            Serial.printf("Updating with new tunings: Kp: %0.2f, Ki: %0.2f, Kd: %0.2f\n",Kp,Ki,Kd);
            break;

        case tuner->runPid: // active once per sample after tunings
            Input = currentTemp;
            myPID->Compute();
            tuner->plotter(Input, Output, Setpoint, 0.1f, 3);
            break;
    }
}
```

`adaptInputSpan()` will check if the set temperature has changed, and if it is, it will re configure the tuner with the new inputSpan. This however did not help on the digital pwm. I added these lines:

```cpp
if( shouldStartHeat && !shouldStartHeatCtr){
    if( millis() - lastShouldStartHeatCtr >= 50000 ){
        shouldStartHeat = false;
        lastShouldStartHeatCtr = millis();
    }
}else if( !shouldStartHeat && shouldStartHeatCtr ){
    if( millis() - lastShouldStartHeatCtr >= 50000 ){
[... 140 lines cut ...]
void adaptInputSpan();
uint32_t settleTimeSec  = 5;
uint32_t testTimeSec    = 5000;
uint16_t samples    = 10000;
float inputSpan     = 350;
float outputSpan    = 1000;
float outputStart   = 0;
float outputStep    = 850;
float tempLimit     = 350;
float prevSetPoint  = 0;

// variables
float Input, Output, Setpoint = 50, Kp, Ki, Kd;
sTune *tuner;
QuickPID *myPID;

void pidInit();
void pidLoop();
/* PID DETAILS */

void checkHeatingStatus();
boolean shouldStartHeat = false;
boolean shouldStartHeatCtr = false;
long lastShouldStartHeatCtr = 0;
```

zekageri, 2023-02-20:
Here is a wokwi which tries to demonstrate my environmnet.

closed 2023-02-21 by zekageri

FelixWT, 2023-02-22:
You closed this issue, so I'm assuming you've got it working now

My apologies, I may have misled you with the wrong coding. The definition of inputSpan should be: 
```
/// @brief Δ °C from Room Temperature to Temperature Limit
float inputSpan;
```
if you look in to to library you can find following equation:
https://github.com/Dlloydev/sTune/blob/94147e1a91036672667f186089614b0db9c525ed/src/sTune.cpp#L194
which is the calculation of process gain, thus inputSpan should be defined as mentioned

---------------------------------------------------------
Little update of my fuction:
```
/// @brief Timer Interrupt Controlled Output
/// @param pidAva PID Available
void sTune::softPwm(bool pidAva, float setpoint, uint32_t windowSize) {
  if(Timer_Event.getTmr0_1ms(windowSize)){
    if(!pidAva){
      snprintf(HeatUpSpan, sizeof(HeatUpSpan),"{\"HOT\":[%.3f]}",*_output);
      Serial.println(HeatUpSpan);
      //Send Relay NC timespan via UART
      Serial2.write(HeatUpSpan);
    }
    OutSpanReach=true;
  }
}
```

zekageri, 2023-02-22:
Thank you for the response. It does not work, i closed this issue to open a new one with my clarifications.
Here is the new issue #29 
PID does not work at all in my environmnet right now.