Dlloydev/sTune #29: PID does not work properly on my environment.

zekageri, 2023-02-21:
**First of all, thank you for this library. It is really a piece of art.**
Probably Iam just **stupid** and can't use it properly.

**My setup**
- ESP32-Wrover-E ( 16mb flash, 8mb psram )
- Arduino environment
- Quick PID
- sTune
- Modbus

The esp32 hosts a webpage where a user can create it's own zones ( rooms ) and can put heating controller inside it.
The user can pick which thermostat to use for which room and can select separate pins for the boiler start, valve open and pump circulation. He can also set how much time it takes for the valve to be opened fully, so the heating can begin and the pumps can circulate the water across the tubes.

**This system is represented as separate classes in code.** 
- A class for the thermostat which gets the measured temperature via modbus from the physical modul and contains the set temperature which a user can controll
- A class for the boiler which will controll the boiler on/off state
- A class for the pump(s) which will controll the pump start/stop and watches for the post circulation time expiration
- A class for the valve(s) which will controll the valve open/close state and watches for the open time expiration
- A heating which will controll the flow of the heating on a particular room

So it goes like this:

Heating loop will periodically checks if we can start to heat or not. For this it will ask the thermostat class which will decide based on the measured and set temperature. If it says we can heat, the heating system will start the process.
- Starts to open the valves
- After the valves are opened ( not known time, can take from 0 to 600sec ) gives a signal to the boiler to start to heat and starts the pumps

Meanwhile the heating system will periodically checks if the temp is on the set temp. If it is, ( +- hysteresis ) it will start the stop process
- Takes away the boiler start signal
- Wait for the defined amount of time to do a post circulation
- After the post circulation done, it will stop the pump(s) and closes the valve(s)

It is working good, but now i want to give the user the option to be able to controll it with PID
For this i have created a wokwi project which tries to simulate my environmnet and for the life of me i can't figure out why it does not work.

Here is the wokwi simulation: [https://wokwi.com/projects/357187661998047233](https://wokwi.com/projects/357187661998047233)

Do you have any suggestions? ( sorry for the long issue )

( meanwhile I will try to create a more representative simulation at wokwi )

FelixWT, 2023-04-06:
Hi, here is some suggestions
1.You should get rid of any process that cause latency while auto tuning , e.g. open valve
2.Take advantage of esp32's duo core . As far as I now setup webserver on esp32 take away lots of efficacy, and might cause extra latency. So by using multi-core you can seperate auto tune process away from websever pooling
3.You should focus on relationsheep between boiler and the sensor , set the pump and valve to a static value when auto tuning

Disclaimer: I'm neither an expert in this field nor very experienced. This is just what I found logical to do.

==============================
p.s. I stopped using this library due to the output algorithm didn't match my needs, but it should works for you since the set point isn't far beyond the initial temperature of your system

zekageri, 2023-04-06:
Thank you for your comment.
The example I gave is non blocking. The valve opening does not use delay and no part of my program blocks. It should not be a computation problem. My system utilises both cores. I squeeze every CPU time from the esp. Valve open time is indeed constant but not at the initialization level. These components are created dynamically. But in my example everything is constant as you can see and it does not work.

FelixWT, 2023-04-06:
I'm afraid there is nothing that I can do for you then.

This blog helps me a lot in understanding some basics of PID
http://brettbeauregard.com/blog/2011/04/improving-the-beginners-pid-introduction/
Here is another place where you can ask for help.
https://groups.google.com/g/diy-pid-control
,good luck

drf5n, 2023-12-27:
> **First of all, thank you for this library. It is really a piece of art.** Probably Iam just **stupid** and can't use it properly.
> 
> **My setup**
> 
> * ESP32-Wrover-E ( 16mb flash, 8mb psram )
> * Arduino environment
> * Quick PID
> * sTune
> * Modbus
> 
> The esp32 hosts a webpage where a user can create it's own zones ( rooms ) and can put heating controller inside it. The user can pick which thermostat to use for which room and can select separate pins for the boiler start, valve open and pump circulation. He can also set how much time it takes for the valve to be opened fully, so the heating can begin and the pumps can circulate the water across the tubes.
> 
> **This system is represented as separate classes in code.**
> 
> * A class for the thermostat which gets the measured temperature via modbus from the physical modul and contains the set temperature which a user can controll
> * A class for the boiler which will controll the boiler on/off state
> * A class for the pump(s) which will controll the pump start/stop and watches for the post circulation time expiration
> * A class for the valve(s) which will controll the valve open/close state and watches for the open time expiration
> * A heating which will controll the flow of the heating on a particular room
> 
> So it goes like this:
> 
> Heating loop will periodically checks if we can start to heat or not. For this it will ask the thermostat class which will decide based on the measured and set temperature. If it says we can heat, the heating system will start the process.
> 
> * Starts to open the valves
> * After the valves are opened ( not known time, can take from 0 to 600sec ) gives a signal to the boiler to start to heat and starts the pumps
> 
> Meanwhile the heating system will periodically checks if the temp is on the set temp. If it is, ( +- hysteresis ) it will start the stop process
> 
> * Takes away the boiler start signal
> * Wait for the defined amount of time to do a post circulation
> * After the post circulation done, it will stop the pump(s) and closes the valve(s)
> 
> It is working good, but now i want to give the user the option to be able to controll it with PID For this i have created a wokwi project which tries to simulate my environmnet and for the life of me i can't figure out why it does not work.
> 
> Here is the wokwi simulation: https://wokwi.com/projects/357187661998047233
> 
> Do you have any suggestions? ( sorry for the long issue )
> 
> ( meanwhile I will try to create a more representative simulation at wokwi )
> 
> Here is a more detailed simulation of my system. https://wokwi.com/projects/357256912224692225 If you let the `usePID` variable `false` it will do a normal heating controll and controlls the temperature fine. If you set it to `true` it will use the `PID` for temp controll.

Looking at the output of the `usePID = false` run shows an constant +0.90°C per observation.  This process doesn't appear to be a self-regulating process which has an inflection point as its S-curve trajectory tapers off towards an ultimate temperature one would reach if you left the system on continuously.  This is an integrating process, which essentially changes by some constant `C` 
 times the ControlOutput per timestep`Ti+1 = Ti + C*CO/dt`.  Maybe the sTune analysis could work on the derivative of the process to determine the shape of the change of the slopes?

```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1156
load:0x40078000,len:11456
ho 0 tail 12 room 4
load:0x40080400,len:2972
entry 0x400805dc
Starting heat controll simulation.

[THERMOSTAT INFO]
- Measured temp: 15.00c°
- Set temp: 23.00c°
- Hysteresis: 0.30c°
[THERMOSTAT INFO]

[VALVE] - Opening.
[VALVE] - Fully opened.
[BOILER] - Starting.
[PUMP] - Starting.
[TEMPERATURE] - 15.90c°
[TEMPERATURE] - 16.80c°
[TEMPERATURE] - 17.70c°
[TEMPERATURE] - 18.60c°
[TEMPERATURE] - 19.50c°
[TEMPERATURE] - 20.40c°
[TEMPERATURE] - 21.30c°
[TEMPERATURE] - 22.20c°
[TEMPERATURE] - 23.10c°
[PUMP] - Starting post circulation.
[PUMP] - Post circulation end.
[PUMP] - Stopped.
[BOILER] - Stopping.
[VALVE] - Closing.
[TEMPERATURE] - 23.00c°
[VALVE] - Opening.
[VALVE] - Fully opened.
[BOILER] - Starting.
[PUMP] - Starting.
[TEMPERATURE] - 23.90c°
[PUMP] - Starting post circulation.
```